uArm-Developer/UArmForArduino #3: Potential issue with CALIBRATION_FLAG

dcorboy, 2016-01-06:
I've noticed that both this firmware package as well as the stock UF_uArm_Metal_new library both use the calibration flag of 0xEE at address 0 of the board EEPROM.

However it appears that the EEPROM calibration data generated by this firmware is a different format, and is stored at a different address in the EEPROM.

Because this calibration method overwrites the entire EEPROM during calibration, it would seem that if a uArm is calibrated by this method, using the original firmware (for record/playback) would result in an calibrated uArm. Further, because of the shared flag, the 3-beeps that warn the user that calibration is needed would not sound.

If this is correct, would it not be good to use a different CALIBRATION_FLAG value for this new format?

UFactory-Alex, 2016-01-08:
Hi, We realized this problem, We should use a new Flag (in different address). Thanks for your kind advice.

UFactory-Alex, 2016-05-06:
New Version with all CALIBRATION_FLAG address

closed 2016-05-06 by UFactory-Alex