uArm-Developer/UArmForArduino #4: Some issues with the new constraints

dcorboy, 2016-01-08:
I have been using the latest code (SHA f09339) and have noticed that the new constraints seem quite restrictive and I wonder if this is intentional.

For example, a move to (4.00, -10.00, 12.00)
results in a final location of (5.46, -15.17, 15.52)
due to the constraints.

Generally, it appears to be this constraint that is most often the reason (line 76, uArm_library.cpp):
if(servo_2_angle_execute + servo_3_angle_execute > 150)

These constraints are even hit during the calibration examples, though the effect is not as severe.

If the constraints are avoided, there can still be some errors of +- 2cm (maybe that should be expected?)

My calibration values seem reasonable:

l_address: 60
servo 1
lr.getA():-29.32
lr.getB():0.35
l_address: 66
servo 2
lr.getA():-27.40
lr.getB():0.34
l_address: 72
servo 3
lr.getA():-30.55
lr.getB():0.35
l_address: 78
servo 4
lr.getA():-43.24
lr.getB():0.50

Rot: 0.95
L: 21.78
r: 2.00

(also, I wonder if lines 72 & 73 (uArm_library.cpp) contain a typo and should be constraining servo 3 instead of servo 2)

This code is very valuable and I hope you will make a formal release soon!

songjunyi, 2016-01-08:
Hi, dcorboy,

This is joey who wrote the original code.

First of all, thanks so much for finding these bugs and we are working on it. 

First  
```
if(servo_2_angle_execute + servo_3_angle_execute > 150)
```
This code will not influence the calibration process. Because there are two writeAngles() function and this is written in the writeAngels(a1,a2,a3,a4). Calibration uses the angle writeAngles(a1,a2,a3,a4,trigger).

Second

Your uarm calibration result seems not as good as we expect. 
```
Rot: 0.95
L: 21.78
r: 2.00
```
In this data, the offset of L is large. So you may calibrate uarm another time may help you with it. It is welcomed if you have any other method to calibrate uArm. 

Third
For
```
also, I wonder if lines 72 & 73 (uArm_library.cpp) contain a typo and should be constraining servo 3 instead of servo 2
```
Thanks for finding this bug and I will fix it asap.
The code for constrains is made for protecting servos.

Thanks so much
Joey

dcorboy, 2016-01-12:
Hi Joey,
Just a note that I've not been able to get the left shoulder servo offset below 20, no matter how carefully I adjust the posture. I have watched the video many times and am sure of the posture.

Could I be doing something wrong? Is it possible the servo was mounted incorrectly? (If so, should I fix it?)

Thanks so much,
Dave

dcorboy, 2016-01-28:
Resolved by ace1702

closed 2016-01-28 by dcorboy