uArm-Developer/UArmForArduino #9: Pump not released immediately

ArifRahmanMY, 2016-01-28:
I noticed that the vacuum pump did not immediately release upon turning it off. [See the problem here.](https://www.youtube.com/watch?v=W4fQk0kfwro)

I have found out that this is caused by the valve being disabled a bit too fast to release all the compression. So I added a bit more delay, and that solved the problem :)

In uarm_library.cpp;
```
void uArmClass::pumpOff()
{
   pinMode(PUMP_EN, OUTPUT);
   pinMode(VALVE_EN, OUTPUT);
   digitalWrite(VALVE_EN, HIGH);
   digitalWrite(PUMP_EN, LOW);
   delay(50);	// <--- changed from 20 to 50
   digitalWrite(VALVE_EN,LOW);
}
```

So I suggest changing this value in the github copy as well. 

Thank you. 

Arif

dcorboy, 2016-01-28:
Hi Arif,

I have noticed this a couple of times but generally use heavier objects where the problem is not as evident.
Let me run a couple of tests and then I can submit a change for the main code.

Dave

dcorboy, 2016-01-30:
I took a look at my own code (that doesn't exhibit the problem) and found that I'd increased the delay as well! I must have had to solve the problem at one point.

I've submitted a pull request (#11) based on your suggestion to address the issue for the main library. Once that's pulled in, we can close this issue.

dcorboy, 2016-01-31:
Joey put in a commit for this e2c6edb. Closing.

closed 2016-01-31 by dcorboy