uArm-Developer/UArmForArduino #14: uArmClass drawRec

anudels, 2016-05-11:
I am trying to run the example program "Test" that comes with the "uArmLibrary" but the current iteration of the uArmClass does not have a drawRec function. Does anyone know where I can get this?

dcorboy, 2016-05-11:
Ah yes. It looks like those demo function were removed with the latest code updates from @UFactory-Alex. I am not sure if they will be replaced as they probably don't belong in the core library.

In any case, I dug up the code for you. I presumed you might be interested in the DrawCur function as well:

```
void uArmClass::drawRec(double length_1, double length_2, double time_spend_per_length)
{
  moveTo(length_1,0,0,1,time_spend_per_length);
  moveTo(0,length_2,0,1,time_spend_per_length);
  moveTo(-length_1,0,0,1,time_spend_per_length);
  moveTo(0,-length_2,0,1,time_spend_per_length);
}

void uArmClass::drawCur(double length_1, double length_2, int angle, double time_spend)
{
  uarm.attachAll();
  double l_xp;
  double l_yp;

  calXYZ();
  double current_x = g_cal_x;
  double current_y = g_cal_y;
  double current_z = g_cal_z;
  double interp_arr[INTERP_INTVLS];

  interpolate(0, angle/MATH_TRANS, interp_arr, INTERP_EASE_INOUT_CUBIC); 

  for (byte i = 0; i < INTERP_INTVLS; i++){

    l_xp = length_1 * cos(interp_arr[i]);
    l_yp = length_2 * sin(interp_arr[i]);

    calAngles( l_xp + current_x - length_1 , l_yp+ current_y , current_z);
    uarm.writeAngle(g_theta_1, g_theta_2, g_theta_3,0);

    delay(time_spend*1000/INTERP_INTVLS);
  }
}
```
These may need a bit of modification, but they should be pretty close to running as-is.

I will leave this issue open until Alex can respond about those functions. If you have more questions on getting your project working, we're happy to answer those over on the forum at https://forum.ufactory.cc/

UFactory-Alex, 2016-05-12:
It's a to-do item, We are planing to move drawRec & drawCur into examples.

closed 2016-05-12 by anudels

anudels, 2016-05-16:
Could someone take a look at the questions that I posted to the forum?

Thank you for your help.