uArm-Developer/UArmForArduino #17: Wrist Servo Timing

ProbablePrime, 2016-08-20:
Looks like there's an incorrect pulse width timing for the wrist servo in the library.

My sketches cause the servo to make whiny noises. All other servos are fine.

The wrist servo is a different servo. Thus these settings are probably different for it?

```
#define MIN_PULSE_WIDTH       500     // the shortest pulse sent to a servo  
#define MAX_PULSE_WIDTH      2500     // the longest pulse sent to a servo 
#define DEFAULT_PULSE_WIDTH  1500     // default pulse width when servo is attached
#define REFRESH_INTERVAL    20000     // minumim time to refresh servos in microseconds 
```

Do you have the model numbers for the servos. Particularly the smaller one.  I can get the details I need from there.

TonyLeheng, 2016-09-10:
We can try 600~2400 for the 9G servo. Make sure the big servos are still 500~2500.
Thanks

TonyLeheng, 2016-09-10:
The pulse width should be 600~2400 for the 9G servo. Make sure the big servos are 500~2500.
Thanks