uArm-Developer/UArmForArduino #20: Arduino crashes when attaching certain servos

apockill, 2016-09-07:
When I do the following:

-------------------------
[sDetS0]
[sDetS1]
[sDetS2]
[sDetS3]
*Stretch the arm as far as it can be stretched*
[sAttS1]
-------------------------

The Arduino crashes: The robot's Buzzer rings, all servos are attached, and then the robot returns back to home position. This also happens in various other positions.

This kind of behavior could severly break uArm Creator Studio.

UFactory-Alex, 2016-09-09:
I rolled back to the previous version, will solve this issue.
But theblow bug still exist.   
If Detach servo and move uArm to a position that `move_to` can not reach. Then Attach back the Servo. It would mess out the coordinate in uArm. Because uArm real range is bigger than the coordinate system range.
For now the best solution is:
when attach back the servo, Move uArm to closest point which can be reached by uArm's coordinate.
We didn't complete it yet.

apockill, 2016-09-18:
Tested it. It's fixed.

closed 2016-09-18 by apockill