uArm-Developer/UArmForArduino #29: (V2.1) Raw vs Offset Input/Output is complicated

apockill, 2016-10-12:
Problem:
We need to simplify the "offset" or "raw" servo angles.

Why: 
There are so many commands for doing the same thing, and they are confusing.
[sSerN#V#]
[sAngN#V#]
[gAng]
[gSer]

It's confusing what commands do what. 

Does [gIKX#Y#Z#] return angles WITH offset or WITHOUT offset? Does [gFKT#L#R#] expect angles with an offset or no offset? 

Solution:
I'm not sure. What do you suggest?

yuntian1019, 2016-10-14:
[gAng] [sAng] return ideal angles but [gSer] [sSer] return raw angles.
idea angle = raw angle + assembly offset
So user should alway use ideal angles.

[gIKX#Y#Z#] and [gFKT#L#R#] return ideal angles(with offset).

apockill, 2016-10-17:
Thanks for clearing that up. I think the documentation for [gIKX#Y#Z#] and [gFKT#L#R#] should mention that they return ideal angles.