uArm-Developer/UArmForArduino #36: Example Calibration just compile is fail.

Minxiangliu, 2017-09-25:
[wrong.docx](https://github.com/uArm-Developer/UArmForArduino/files/1329985/wrong.docx)
Hello,I download your uArm-Developer library and and install to Arduino IDE.
I need Calibration the uArm Metal.
Open the Calibration Example code no change anything just compile.
I use Arduino/Genuino Uno to compile.
The results is many wrong.
Many wrong I not understand but some of them are about multiple definition.
How can i improve it?
I have uploaded an error message.
thank you.

dukelec, 2017-09-25:
Try to move "libraries/Servo" to another place other than libraries/.

Minxiangliu, 2017-09-26:
To dukelec:
You mean is ardunio original library of Servo move to another place?
If this is the case,compile also fail.
Because Calibration example have include the Servo.h.
An error message will appear after compiling:
fatal error: Servo.h: No such file or directory

Can you Clearly express what you mean?
thank you.

DickoAI, 2024-01-24:
Hey @Minxiangliu , did you resolve this issue? If so, please can you explain how.