uArm-Developer/UArmForArduino #37: Move example on uArm swift pro?

Meladoo, 2018-05-04:
Hello there,
Do you know how can I perform a similar code to your move example on the uArm Swift Pro? I need to move the swift pro but can not manage.

dcorboy, 2018-05-07:
Are you trying to port the movement code itself, as a library to be used by uArm Swift Pro?

The Metal code is not very portable and some of the math is based specifically on the geometry of the Metal. If I were doing the port, I would copy the code over and then begin translating the parts that are specific to Metal over to Swift Pro versions.

Meladoo, 2018-05-07:
Yes, I tried that. The problem is I am struggling to find the swift pro version methods/commands that will accommodate with this code. (espically service.setButtonService and moveto which seems to be removed) 
Thanks anyway.

dcorboy, 2018-05-07:
Hmm, I don't know the Swift Pro but I would assume that there is a basic
library that lets you move the arm? There must be some examples that show
you how to do basic moves of the servos? With that, you should be able to
translate. Harder part will be the geometry computation, but let me know if
you have the ability to simple write the Swift Pro servos.

Meladoo, 2018-05-07:
Quick look into the library and I found the servos code. 
These are the links: 

https://github.com/uArm-Developer/SwiftProForArduino/blob/develop/lib/Marlin/uArmServo.h
https://github.com/uArm-Developer/SwiftProForArduino/blob/develop/lib/Marlin/uArmServo.cpp

Sorry that I am not an Arduino expert, but will I be able to move the swift pro by putting the servo command in place of the moveTo ?

dcorboy, 2018-05-07:
Yes, those files (and others) would allow you to recreate something like the uArm Metal functionality using Swift Pro.

However, it is a bigger task than you may think. Metal has the notion of a 'calibration' that Swift Pro does not seem to use. This, and some other details suggest that you'll need to really examine how the current Swift Pro servo code (Marvin) is working before you start writing directly to servos (they can EASILY be burned out with bad programming, I've done it! :)

I don't know if the Swift Pro has the same arm geometry as Metal. If not, all the mathematics of computing position from servo angle (and vice versa) will change.

Finally, since I also assume that the Arduino and pinouts change between the models, all of those changes would need to be done as well.

That said, it would be a great thing for Swift Pro experimenters to have (which is why I did it for Metal). And most of the cool stuff is more portable and wouldn't need modification.